khoih-prog/WiFiNINA_Generic #1: u-blox nina b

tcpipchip, 2020-04-21:
@khoih-prog 

Hi

Is it possible to port to NRF52 chips ? I can test here! 

Are the routines HIGH level ? (hardware gpios access)

Thanks!

khoih-prog, 2020-04-21:
Hi,
I currently have no plan yet to port to other boards such as Nano 33 BLE / BLE Sense.

> Is it possible to port to NRF52 chips ?

Do you mean the board using nRF52x such as Nano 33 BLE / BLE Sense (using nRF52840) or just the uBlox NINA B1/2/3/4 modules (using nRF52832, etc.)?

> I can test here!

Great and appreciated, thanks for your offer to help. 
Still waiting for your answers before investigating to see what are already available and not, then acquiring modules/boards and deciding if and what to do.

tcpipchip, 2020-04-21:
Hi, to NINA B302 that is a NRF52840.
I am the support to Smartcore.com.br, the U-BLOX representative in Brazil! I want to offer to their clients (that uses the NINA B302) the option to add the WIFI stack!
I want to do a blog about and give you the merits!

Just now i got success to add WIFI to NINA B302, but using the PYTHON. But, i am looking for ARDUINO

If you make work with 33 BLE SENSE, it will work with NINA B302 !

Thanks!

khoih-prog, 2020-04-21:
> Just now i got success to add WIFI to NINA B302, but using the PYTHON.

That's great. Could you pass the link or any additional info so that I can have a look and save some time researching.

> If you make work with 33 BLE SENSE, it will work with NINA B302 !

Will try.

Thanks,

tcpipchip, 2020-04-21:
Yes
https://www.youtube.com/watch?time_continue=103&v=U-xRKa_7iro&feature=emb_logo 
https://www.rototron.info/circuitpython-nrf52840-esp32-wifi-tutorial/

khoih-prog, 2020-04-21:
Thanks for the links.

I'll try to make WiFi running on Nano 33 BLE (nRF52840). Will let you know if it's OK soon.

tcpipchip, 2020-04-21:
@khoih-prog 

This is the best news of today!!!!!

And you will be famous :) too

I am will write many blogs with ble and wifinina with your credits!

khoih-prog, 2020-04-21:
> And you will be famous :) too
I really don't like that as being obscure will help have time to do better work.
I'm doing this just to spend my remaining free time to help the community and to make my brain not rusty. NO further thinking.
But many thanks for your nice words anyway.

**PS:** 
1. The W302 in Nano 33 BLE, I think, can only run either BLE or WiFi, **not both at the same time** with the current firmware as there is only 1 radio channel. The better  [U-Blox NINA W15](https://www.u-blox.com/en/product/nina-w15-series) can do this simulltaneously
2. It's great if you, as one of UBlox reps, can ask UBlox to **upgra.de the firmware** to let both WiFi and BLE running at the same time as in Nano 33 IoT. If this is possible, the work will be much simpler then.
3. Another question is why you don't use **Nano 33 Iot which offer both WiFi and BLE (slave) at the same time**?

tcpipchip, 2020-04-22:
And you will be famous :) too
I really don't like that as being obscure will help have time to do better work.
I'm doing this just to spend my remaining free time to help the community and to make my brain not rusty. NO further thinking.
But many thanks for your nice words anyway.

YES, YOU ARE RIGHT!

PS:

The W302 in Nano 33 BLE, I think, can only run either BLE or WiFi, not both at the same time with the current firmware as there is only 1 radio channel. The better U-Blox NINA W15 can do this simulltaneously

BUT DO YOU HAVE ONE WITH W102 ? LOOKS THAT W302 HAS 4MB FLASH!

It's great if you, as one of UBlox reps, can ask UBlox to upgra.de the firmware to let both WiFi and BLE running at the same time as in Nano 33 IoT. If this is possible, the work will be much simpler then.

BUT THAT FIRMWARE OF MODULE IS NOT OF UBLOX TEAM!

Another question is why you don't use Nano 33 Iot which offer both WiFi and BLE (slave) at the same time?

UBLOX REPRESENTATIVE ONLY HAVE W102,B302 and B112 and B312

tcpipchip, 2020-04-22:
https://www.blogger.com/profile/01246279301136130630

khoih-prog, 2020-04-22:
> BUT DO YOU HAVE ONE WITH W102 ?

I have Nano 33 IoT using U-blox W102. All the libraries relating to WiFiNINA are written for and tested specially on that Nano board.

> BUT THAT FIRMWARE OF MODULE IS NOT OF UBLOX TEAM!

So strange, at least they have some engineers supporting / developing these firmwares with Espressif (ESP-32 based chip). Otherwise, it'll take much more time for outsiders to do that.
I think it's also possible to change the firmware of W302 to something similar to ESP32's to support both WiFi and BLE simultaneously.

It'll take some more research before knowing the time, if and when the work can be done. Hopefully  someone will be there much earlier.

tcpipchip, 2020-04-22:
Nano 33 iot uses atmel SAM

The challenger is replace the SAM by a NRF52 :)

About u-blox, they have proprietary firmware to W1 (for the chips that are not opencpu)

khoih-prog, 2020-04-22:
Hi,
Your blog is very informative and good.

After some research, I have these following observations:
1. The [WiFi Tutorial for nRF52840](https://www.rototron.info/circuitpython-nrf52840-esp32-wifi-tutorial/) is not about using W302 to run WiFi, but using external ESP32 as co-processor to run WiFi.
2. It's better to use **W102**, W1x if you'd like to run WiFi as in **Nano 33 IoT**, instead of using W302, W3x (as in **Nano 33 BLE/Sense**).
3. The **Arduino mbed-based** development (for Nano 33 BLE/Sense) is still pre-matured (without the Network, WiFi, etc. support). It'll take some much more time to wait or develop yourself.

IMHO, the suggestion and plan is:
1. Don't go further with the WiFiNINA port to **W302**, etc.
2. Port the WiFiNINA ( [WiFiWebServer](https://github.com/khoih-prog/WiFiWebServer),  [WiFiNINA_Generic](https://github.com/khoih-prog/WiFiNINA_Generic), [WiFiManager_NINA_Lite](https://github.com/khoih-prog/WiFiManager_NINA_Lite), [Blynk_WiFiNINA_WM](https://github.com/khoih-prog/Blynk_WiFiNINA_WM) ) libraries to **nRF52 boards** (AdaFruit  Feather nRF52832, nRF52840 Express, BlueFruit Sense, Itsy-Bitsy nRF52840 Express, Metro nRF52840 Express, etc.).
3. Enable the above-mentioned AdaFruit boards to run **Ethernet** W5x00, ENC28J60, LAN8742A, etc. Will buy some boards to test.
4. **Enhance** Arduino mbed-based development (for Nano 33 BLE/Sense) to add WiFi/Ethernet Network support. Some long-term plan for fun if not available then.

Hope they are correct after so short research time .

tcpipchip, 2020-04-22:
late i will try to make the NRF52 try to talk with W102 (WIFININA) too! But i still testing it with CIRCUIT PYTHON and need to make some blogs

khoih-prog, 2020-04-22:
> i will try to make the NRF52 try to talk with W102 (WIFININA)

Pls inform when done. I can port the WiFiNINA libraries to nRF52 boards, but still don't have any board (to be ordered) to test. Hopefully you can help test  when the enhancement finished.

tcpipchip, 2020-04-22:
Hi,
Your blog is very informative and good.

**Thanks!!!! Helps the community!**

After some research, I have these following observations:

The WiFi Tutorial for nRF52840 is not about using W302 to run WiFi, but using external ESP32 as co-processor to run WiFi.

**Yes, the W102!**

It's better to use W102, W1x if you'd like to run WiFi as in Nano 33 IoT, instead of using W302, W3x (as in Nano 33 BLE/Sense).

**yes, i want the W102**

The Arduino mbed-based development (for Nano 33 BLE/Sense) is still pre-matured (without the Network, WiFi, etc. support). It'll take some much more time to wait or develop yourself.
IMHO, the suggestion and plan is:

**I never more used the Mbed, but they are making changes on Web page and OS**

Don't go further with the WiFiNINA port to W302, etc.
Port the WiFiNINA ( WiFiWebServer, WiFiNINA_Generic, WiFiManager_NINA_Lite, Blynk_WiFiNINA_WM ) libraries to nRF52 boards (AdaFruit Feather nRF52832, nRF52840 Express, BlueFruit Sense, Itsy-Bitsy nRF52840 Express, Metro nRF52840 Express, etc.).

Yes!!!! Make NRF528XX (Arduino) talks with W102 (WIFININA)

Enable the above-mentioned AdaFruit boards to run Ethernet W5x00, ENC28J60, LAN8742A, etc. Will buy some boards to test.

**that is a good idea!!! It will make NRF52 for example, a NRF52 MESH reach WAN :)**

Enhance Arduino mbed-based development (for Nano 33 BLE/Sense) to add WiFi/Ethernet Network support. Some long-term plan for fun if not available then.
Hope they are correct after so short research time .

**Correct!**

tcpipchip, 2020-04-22:
[image]

Left side is B302 (NRF52840) running CIRCUITPYTHON and right side is W102 running WIFININA, both connected with SPI

tcpipchip, 2020-04-22:
[image]

tcpipchip, 2020-04-22:
Try now make a NRF52 works with W102 :)

khoih-prog, 2020-04-22:
Just finish porting WiFiNINA_Generic to AdaFruit nRF52 boards. Will post today as v1.5.1 for you to test as I don't have the board to test.
It's now supporting nRF52 boards running WiFiNINA W102 shields.
Could you please prepare to test.

tcpipchip, 2020-04-22:
Good night! you are so FAST!!!!!!!!!!!!!!!!!!!!!!!!!!!!

Yes, i can test!!!!!!!!!!!

Good news!

tcpipchip, 2020-04-22:
Please, give some time to do that! I am going to the classroom in a few minutes

tcpipchip, 2020-04-22:
@khoih-prog 
Did you compile in your side and got errors during the compile ?
https://1drv.ms/t/s!Ak_-1v7Ug-k28UoIb2Vk8dveagDQ
[image]

tcpipchip, 2020-04-22:
What is the  v1.5.1  that you said ?
New examples ?

khoih-prog, 2020-04-22:
Pls wait for v1.5.1 in an hour. Don't use v1.5.0 for this purrpose.
I'm doing final check before posting now.

khoih-prog, 2020-04-22:
It's ready now, v1.5.1

Compile result for WiFiWebServer.ino:

[image]

khoih-prog, 2020-04-22:
Waiting for the test result to continue to update the following libs to support nRF52 boards:

1. [**WiFiWebServer**](https://github.com/khoih-prog/WiFiWebServer),
2. [**WiFiManager_NINA_Lite**](https://github.com/khoih-prog/WiFiManager_NINA_Lite),
3. [**Blynk_WiFiNINA_WM**](https://github.com/khoih-prog/Blynk_WiFiNINA_WM) 
4. [**EthernetWebServer**](https://github.com/khoih-prog/EthernetWebServer)
5. [**BlynkEthernet_WM**](https://github.com/khoih-prog/BlynkEthernet_WM)

tcpipchip, 2020-04-23:
i got 3 minutes now (i am in classroom, and compiles)! Tomorrow i will test after change MAP pins!

Help me to understand the #defines

#define NINA_GPIO0  (26u)                             //26
 #define NINA_RESETN (27u)
 #define NINA_ACK    (28u)

  #define SPIWIFI_SS       24   //PIN_SPI1_SS            //24
  #define SPIWIFI_ACK      28   //NINA_ACK               //28 
  #define SPIWIFI_RESET    27   //NINA_RESETN            //27

[image]

in Circuit Python i choose that pins (left side), P0.X, in the NRF52 side, not clear to me wich are the simular in your defines :(

khoih-prog, 2020-04-23:
> What is the function of
> #define NINA_GPIO0 ?

I have to check more to see how the pin is used.

You can see how the pins are used in Nano 33 IoT as follows:

[image]

and the corresponding **#defines**

//NINA pin to control
#define NINA_GPIO0  (26u)       <==== NiNA W102 GIPIO0   <--> pin 26 of SAMD21
#define NINA_RESETN (27u)    <==== NiNA W102 RESETN <--> pin 27 of SAMD21   
#define NINA_ACK    (28u)        <==== NiNA W102 ACK         <--> pin 28 of SAMD21

// SPI1 pins for SPI bus communications
#define PIN_SPI1_MISO    (23u)  <==== NiNA W102 MISO   <--> pin 23 MISO of SAMD21
#define PIN_SPI1_MOSI    (22u)  <==== NiNA W102 MOSI   <--> pin 23 MOSI of SAMD21
#define PIN_SPI1_SCK     (25u)   <==== NiNA W102 SCK    <--> pin 25 SCK of SAMD21
#define PIN_SPI1_SS      (24u)     <==== NiNA W102 SS/CS <--> pin 24  of SAMD21

[image]

???       <==== NiNA W102 GIPIO0   <--> pin 26 of SAMD21
Pin 19 of W102  <==== NiNA W102 RESETN <--> pin 27 of SAMD21   
???        <==== NiNA W102 ACK         <--> pin 28 of SAMD21

// SPI1 pins for SPI bus communications
Pin 21 of W102  <==== NiNA W102 MISO   <--> pin 23 MISO of SAMD21
Pin 1 of W102    <==== NiNA W102 MOSI   <--> pin 23 MOSI of SAMD21
Pin 29 of W102  <==== NiNA W102 SCK    <--> pin 25 SCK of SAMD21
Pin 28 of W102   <==== NiNA W102 SS/CS <--> pin 24  of SAMD21

> where in the source code i define that i am using of these
> ( defined(NRF52840_FEATHER) || defined(NRF52832_FEATHER) || defined(NRF52_SERIES) || defined(ARDUINO_NRF52_ADAFRUIT) )

You don't need to do anything unless you'd like to print **board information** depending on the board you're using. The compiler will take care of that. Just select the correct board in Arduino Board Manager.

Have a look at the example [WiFiWebServer](https://github.com/khoih-prog/WiFiNINA_Generic/tree/master/examples/WiFiWebServer)

If you like to print board type, use
```
#if    ( defined(NRF52840_FEATHER) || defined(NRF52832_FEATHER) || defined(NRF52_SERIES) || defined(ARDUINO_NRF52_ADAFRUIT) )
  #if defined(WIFI_USE_NRF528XX)
    #undef WIFI_USE_NRF528XX
  #endif
  #define WIFI_USE_NRF528XX          true

  #if defined(NRF52840_FEATHER) 
  #define BOARD_TYPE      "NRF52840_FEATHER"
  #elif defined(NRF52832_FEATHER)
  #define BOARD_TYPE      "NRF52832_FEATHER"
  #else
  #define BOARD_TYPE      "NRF52 Unknown"
  #endif
#endif
....

....
void setup(void)
{
  Serial.begin(115200);
  while (!Serial);
  
  Serial.println("\nStarting AdvancedServer on " + String(BOARD_TYPE));
  ....
}
...
```

khoih-prog, 2020-04-23:
For W302 <--> nRF52840, I think you can connect as follows:

??? <==== NiNA W302 GIPIO0 <--> pin ?? of nRF52840
Pin GPIO_3 of W302 <==== NiNA W302 RESETN <--> pin PO.15 of nRF52840
Pin GPIO_2 of W302 <==== NiNA W302 ACK <--> pin PO.14 of nRF52840

// SPI1 pins for SPI bus communications
Pin GPIO_5 of W302 <==== NiNA W302 MISO <--> pin PO.24 MISO of nRF52840
Pin GPIO_7 of W302 <==== NiNA W302 MOSI <--> pin PO.25 MOSI of nRF52840
Pin GPIO_4 of W302 <==== NiNA W302 SCK <--> pin PO.16 SCK of nRF52840
Pin GPIO_1 of W302 <==== NiNA W302 SS/CS <--> pin PO.13 of nRF52840

tcpipchip, 2020-04-23:
I did all the Setup, still nothing running! No DEBUG message showing what is going on!

Maybe the problem is that GIPIO0 on W102, strange that i didnt use this GPIO on CIRCUIT PYTHON...

khoih-prog, 2020-04-23:
Which boards are you testing now? Can you send the actual pin-to-pin connections as well as any terminal debugging msg.

Adafruit Feather nRF52840 Express? W102 or W302?
I can put more debugging msgs to see what's happening and put on master and let you know when.

BTW, can you swap the MOSI and MISO pins.

By the times, I'll have a look at GPIO0 to see what its actual function.

tcpipchip, 2020-04-23:
Hi
Left side, NINA B302 (NRF52840), right side, NINA W102 (ESP32)
Tomorrow i send you their connection used on CIRCUIT PYTHON SOFTWARE!
Please, put messages debug!
I created a FEATHER variant, the NINA_B302_Ublox, works nice with gpio access to arduino!
I had already swap the miso mosi

Debug messages is a good begin!

I am in the bed :)

Enviado do meu iPhone

khoih-prog, 2020-04-23:
The Nano 33  IoT schematics shows how W102 is connected

[image]

**Pls discard the previous pin-to-pin connections and use the following:**

[image]

Note:  NiNA W102 GIPIO0 pulled low during startup for download
firmware. GPIO0 pulled UP then DOWN for Normal Boot from Internal Flash

**Pin 27 of W102  <==== NiNA W102 GIPIO0** <--> pin PA27 of SAMD21 (100K PULL_UP)
Pin 19 of W102 <==== NiNA W102 RESETN <--> pin PA8 of SAMD21
Pin 7 of W102 <==== NiNA W102 ACK <--> pin PA28 of SAMD21

// SPI1 pins for SPI bus communications
Pin 21 of W102 <==== NiNA W102 MOSI <--> pin PA12 MOSI of SAMD21
Pin 1 of W102 <==== NiNA W102 MISO <--> pin PA13 MISO of SAMD21
**Pin 29 and 20 of W102 <==== NiNA W102 SCK** <--> pin PA15 SCK of SAMD21
**Pin 28 and 21 of W102 <==== NiNA W102 SS/CS** <--> pin PA14 of SAMD21

The GPIO0 function in W102

```
void SpiDrv::begin()
{
...
      SPIWIFI.begin();
      pinMode(SLAVESELECT, OUTPUT);
      pinMode(SLAVEREADY, INPUT);
      pinMode(SLAVERESET, OUTPUT);
      
      // The so-called W102 GPIO0 is **SYS_BOOT**/GPIO_27/RMII_CLK
      pinMode(NINA_GPIO0, OUTPUT);  // Pin is output
      
     // GPIO0 pulled UP then DOWN for Normal Boot from Internal Flash
      digitalWrite(NINA_GPIO0, HIGH);   // Pulse HIGH
      digitalWrite(SLAVESELECT, HIGH);  // then W102's CS HIGH
      digitalWrite(SLAVERESET, inverted_reset ? HIGH : LOW);  // reset W102 for 10ms
      delay(10);
      digitalWrite(SLAVERESET, inverted_reset ? LOW : HIGH); // reset W102 done
      delay(750);  // wait 750ms

      digitalWrite(NINA_GPIO0, LOW);   // pulse GPIO0 LOW
      pinMode(NINA_GPIO0, INPUT);   // make GPIO0 INPUT now. 
```
Another question is if you updated the firmware of W102 to latest v1.3.0 ?

khoih-prog, 2020-04-23:
_Already post in Master the debugging version._

I'm afraid that the **B302 firmware** you have with the board is not suitable for running the normal program as normal nRF52840. Do you have an Adafruit Feather nRF52840 Express (instead of B302) to test to eliminate? I'm not sure the SPI bus on B302 board is working OK as expected (you can check MISO and MOSI signals to see).

The Adafruit assignments for MOSI, MISO and CS/SS are also different (in variants/xxx/variant.cpp) for every board, depending which we choose when compile :

1) **Feather nRF52840 Express**
// D24 .. D26 (aka SPI pins)
  15,  // D24 is P0.15 (SPI MISO)
  13,  // D25 is P0.13 (SPI MOSI)
  14,  // D26 is P0.14 (SPI SCK )

2) **Itsy-Bitsy nRF52840 Express**
// D23 .. D25 (aka SPI pins)
  20,  // D23 is P0.20 (SPI MISO)
  15,  // D24 is P0.15 (SPI MOSI)
  13,  // D25 is P0.13 (SPI SCK )

3) **Metro nRF52840 Express**
  _PINNUM(0, 11), // D24 is SPI MISO on P0.11
  _PINNUM(1, 8 ), // D25 is SPI MOSI on P1.08
  _PINNUM(0, 7 ), // D26 is SPI SCK on P0.07

The problem is I don't have any nRF52 board here to test, as I just ordered some from Adafruit today and it'll take longer time to arrive in this nCovid condition.

Just try out best now.
To connect W102 to B302 if you select **Feather nRF52840 Express** to compile:

[image]

The actual pin assignment will be:
```
Pin 27 of W102 <==== NiNA W102 GIPIO0 <--> pin 42/GPIO_42/P0.26 of nRF52840
Pin 19 of W102 <==== NiNA W102 RESETN <--> GPIO_7/P0.25 of nRF52840  (100K PULL_UP)
Pin 7 of W102 <==== NiNA W102 ACK <--> /GPIO_5/P0.24 of nRF52840

// SPI1 pins for SPI bus communications
Pin 21 of W102 <==== NiNA W102 MOSI <--> D25/P0.13 MOSI of nRF52840
Pin 1 of W102 <==== NiNA W102 MISO <-->D24/P0.15 MISO of nRF52840
Pin 29 and 20 of W102 <==== NiNA W102 SCK <--> D26/P0.14 SCK of nRF52840
Pin 28 and 21 of W102 <==== NiNA W102 SS/CS <--> GPIO_4/P0.16 of nRF52840
```
Change the pin in WiFiNINA_Pinout_Generic.h according to actual connection between W102 and nRF52 to:
```
#elif ( defined(NRF52840_FEATHER) || defined(NRF52832_FEATHER) || defined(NRF52_SERIES) || defined(ARDUINO_NRF52_ADAFRUIT) )

#warning You have to modify pin usage accoring to actual connection for NRF528XX
// To define pin out for WiFiNINA here

//#define PINS_COUNT (60u)
//NINA
#define NINA_GPIO0 (26u)        //P0.26
#define NINA_RESETN (25u)       //P0.25
#define NINA_ACK (24u)          //P0.24

#define SPIWIFI_SS           16 //PIN_SPI1_SS //P0.16
#define SPIWIFI_ACK          24 //NINA_ACK //P0.24
#define SPIWIFI_RESET        25 //NINA_RESETN //P0.25
```

khoih-prog, 2020-04-23:
Hi,
I've created a private repository [WiFiNINA_Generic_Dev](https://github.com/khoih-prog/WiFiNINA_Generic_Dev) for us to work on to avoid possibly lengthy and unnecessary communications to other people.
When everything is OK, we'll merge back to this public repository.
Already invited you in. I'll move these posts here to that [WiFiNINA_Generic_Dev repository](https://github.com/khoih-prog/WiFiNINA_Generic_Dev)  as well.

tcpipchip, 2020-04-23:
HI @khoih-prog 

just wake up! I will do the changes and tests in 2,5 hours! Going to the VIRTUAL classroom!

Thanks!!!!

khoih-prog, 2020-04-23:
Hi @tcpipchip 

I'm temporarily closing this issue now. Pls post on private repository

[u-blox nina b](https://github.com/khoih-prog/WiFiNINA_Generic_Dev/issues/1)

closed 2020-04-23 by khoih-prog

khoih-prog, 2020-04-23:
**From tcpipchip**

GOOD NEWS!!!!!!!!!!!!!!!!!
[image]

[image]

CONGRATULATIONS FOR US!!!!!!!!!!!!!!!!!!!

khoih-prog, 2020-04-23:
Finally, thanks to the collaboration of @tcpipchip, the nRF52 (even B302's nRF52840) boards can run WiFiNINA W102 from version v1.5.1-final
Cheers,

khoih-prog, 2020-04-29:
A success story how to use ESP32-based WiFiNINA with Teensy 4.0

[WiFiNINA_Generic-Library-for-WiFiNINA-modules-to-support-Teensy-and-more-boards](https://forum.pjrc.com/threads/60245-WiFiNINA_Generic-Library-for-WiFiNINA-modules-to-support-Teensy-and-more-boards?p=237320&posted=1#post237320)

> Compiling Adafruit NINA-W102 firmware for an ESP32:
> (Note that it's a fork of the Arduino firmware, this all may work with that as well)
> 
> Clone/download this repo:
> https://github.com/adafruit/nina-fw
> The instructions are basically in the readme in that repo.
> 
> At the time of writing it's designed to work with the ESP-IDF v3.3.1, which has instructions of how to set that up here:
> https://docs.espressif.com/projects/...etup-toolchain
> 
> I'm using a Adafruit HUZZAH32 ESP32 Breakout Board which has all the ESP32 pins available, so it seemed to make sense to use the standard SPI pins (which the Adafruit and Arduino versions of the firmware don't seem to do). I changed the pins that the ESP32 uses for SPI communication to those specified as VSPI/SPI3 in the ESP-IDF documentation here:
> https://docs.espressif.com/projects/...rix-and-io-mux
> 
> The pins are:
> CS: 5
> MOSI: 23
> MISO: 19
> SCK: 18
> BUSY: 33
> 
> So the line at the bottom of SPIS.cpp (https://github.com/adafruit/nina-fw/.../SPIS.cpp#L116) is now:

```cpp
SPISClass SPIS(VSPI_HOST, 1, 23, 19, 18, 5, 33);
```

> I used 'make menuconfig' (which is a tool that edits the sdkconfig file) to enable bootloader logging and console UART output, which ended up setting these options in sdkconfig:
> CONFIG_LOG_BOOTLOADER_LEVEL_INFO=y
> CONFIG_LOG_BOOTLOADER_LEVEL=3
> CONFIG_CONSOLE_UART_DEFAULT=y
> CONFIG_CONSOLE_UART_BAUDRATE=115200
> 
> Then run 'make firmware' to build. I uploaded it with a FTDI serial cable using the instructions for esptool.py described here:
> https://learn.adafruit.com/upgrading...h-esptool-3-24