khoih-prog/WiFiNINA_Generic #4: Storage class

AppsByDavideV, 2020-07-28:
Hello,
Your library is great, but the storage part does not work, The original WiFinina does.
It looks something related to the WiFi_Drv class.

`C:\Users\vdavi\AppData\Local\Temp\arduino_build_453956\sketch\Server.ino.cpp.o: In function `setup':

C:\Users\vdavi\Documents\Arduino\libraries\WiFiNINA_Generic\src/utility/wifi_drv.h:344: undefined reference to `WiFiDrv::fileOperation(unsigned char, char const*, unsigned char, unsigned long, unsigned char*, unsigned long)'

C:\Users\vdavi\Documents\Arduino\libraries\WiFiNINA_Generic\src/utility/wifi_drv.h:337: undefined reference to `WiFiDrv::fileOperation(unsigned char, char const*, unsigned char, unsigned long, unsigned char*, unsigned long)'

C:\Users\vdavi\AppData\Local\Temp\arduino_build_453956\libraries\WiFiNINA_Generic\WiFiStorage_Generic.cpp.o: In function `WiFiStorageClass::exists(char const*, unsigned long*)':

C:\Users\vdavi\Documents\Arduino\libraries\WiFiNINA_Generic\src\utility/wifi_drv.h:344: undefined reference to `WiFiDrv::fileOperation(unsigned char, char const*, unsigned char, unsigned long, unsigned char*, unsigned long)'

collect2.exe: error: ld returned 1 exit status`

Maybe you could have a look into that.
Thanks a lot

Davide

khoih-prog, 2020-07-28:
@AppsByDavideV 

Thanks for using the library and encouraging words.

The Storage class is new and I really haven't got time to verify and test the feature. It's good that you found out the issue anf I'll certainly investigate and fix the bug.

Could you pls help by posting the Server.ino sketch to save me time to recreate the issue to fix the bug.

Regards,

khoih-prog, 2020-07-28:
@AppsByDavideV 

I see the issue now. Will fix now and post onto Master, and new v1.6.2 and inform you.

khoih-prog, 2020-07-29:
Hi @AppsByDavideV 

The new [WiFiNINA_Generic v1.6.2](https://github.com/khoih-prog/WiFiNINA_Generic/releases/tag/v1.6.2) fixes the issue.

You can use [WiFiStorage example](https://github.com/khoih-prog/WiFiNINA_Generic/tree/master/examples/WiFiStorage) to test and verify.

This is the debug terminal output when running the  [WiFiStorage example](https://github.com/khoih-prog/WiFiNINA_Generic/tree/master/examples/WiFiStorage)  on SAMD_NANO_33_IOT

```
Start WiFiStorage on SAMD_NANO_33_IOT
WiFiStorage on SAMD_NANO_33_IOT is OK if you see this long line.
```

Your contribution was noted in [Contributions-and-Thanks](https://github.com/khoih-prog/WiFiNINA_Generic#contributions-and-thanks)

closed 2020-07-29 by khoih-prog