khoih-prog/WiFiNINA_Generic #5: Wifistorage class: file.write() error

AppsByDavideV, 2020-08-06:
Hello, just to let you know I opened an issue on the original wifinina project regarding the file.write() error that writes bad data in the ESP32 flash:
Please have a look here:

https://github.com/arduino-libraries/WiFiNINA/issues/125

Thanks :)

khoih-prog, 2020-08-06:
Can you retest using this modified example

[WiFiStorage](https://github.com/khoih-prog/WiFiNINA_Generic/tree/master/examples/WiFiStorage)

to see if there still is the same problem. I think that's just the issue of the original example only.

```
#include <WiFiNINA_Generic.h>

#ifndef BOARD_NAME
  #define BOARD_NAME    "Unknown board"
#endif

void setup() 
{
  Serial.begin(115200);
  while (!Serial);

  Serial.println("\nStart WiFiStorage on " + String(BOARD_NAME));

  // check for the WiFi module:
  if (WiFi.status() == WL_NO_MODULE)
  {
    Serial.println("Communication with WiFi module failed!");
    // don't continue
    while (true);
  }

  String fv = WiFi.firmwareVersion();
  
  if (fv < WIFI_FIRMWARE_LATEST_VERSION)
  {
    Serial.println("Please upgrade the firmware");
  }

  WiFiStorageFile file = WiFiStorage.open("/fs/testfile");

  if (file) 
  {
    file.erase();
  }
  else
    Serial.println("WiFiStorage error");

  String test = "WiFiStorage on " + String(BOARD_NAME) + " is OK if you see this long line.";
    
  file.write(test.c_str(), test.length());

  if (file) 
  {
    uint8_t buf[128];

    file.seek(0);
    file.read(buf, 128);
          
    Serial.println((char*) buf);
  }  
}

void loop() 
{
  // put your main code here, to run repeatedly:
}
```

AppsByDavideV, 2020-08-06:
Hi, thanks for reply so fast. I tried your example but still the same problem. To see the issue you must write some line consecutively (can be Strings or array of chars....) . Most of the time I have some bytes, at the end of a written string, set to 0.

I know it is not a problem inside your library, it is present also in the original one. No matter the example code.
I think it is something related to write timing or so.
It could also be an internal flash writing problem on my Arduino wifi1010 only, I don't know.

Anyway find attached Your example modified to write consecutive lines to the flash., if you want to test on your chip.

//Davide

[flashTest.zip](https://github.com/khoih-prog/WiFiNINA_Generic/files/5036924/flashTest.zip)

khoih-prog, 2020-08-06:
I don't see the issue on Nano-33-IoT. Possibly some library issue with your WiFi 1010

```
Start WiFiStorage on SAMD_NANO_33_IOT
WiFiStorage on SAMD_NANO_33_IOT is OK if you see this long line.

Char: 0:W:87
Char: 1:i:105
Char: 2:F:70
Char: 3:i:105
Char: 4:S:83
Char: 5:t:116
Char: 6:o:111
Char: 7:r:114
Char: 8:a:97
Char: 9:g:103
Char: 10:e:101
Char: 11: :32
Char: 12:o:111
Char: 13:n:110
Char: 14: :32
Char: 15:S:83
Char: 16:A:65
Char: 17:M:77
Char: 18:D:68
Char: 19:_:95
Char: 20:N:78
Char: 21:A:65
Char: 22:N:78
Char: 23:O:79
Char: 24:_:95
Char: 25:3:51
Char: 26:3:51
Char: 27:_:95
Char: 28:I:73
Char: 29:O:79
Char: 30:T:84
Char: 31: :32
Char: 32:i:105
Char: 33:s:115
Char: 34: :32
Char: 35:O:79
Char: 36:K:75
Char: 37: :32
Char: 38:i:105
Char: 39:f:102
Char: 40: :32
Char: 41:y:121
Char: 42:o:111
Char: 43:u:117
Char: 44: :32
Char: 45:s:115
Char: 46:e:101
Char: 47:e:101
Char: 48: :32
Char: 49:t:116
Char: 50:h:104
Char: 51:i:105
Char: 52:s:115
Char: 53: :32
Char: 54:l:108
Char: 55:o:111
Char: 56:n:110
Char: 57:g:103
Char: 58: :32
Char: 59:l:108
Char: 60:i:105
Char: 61:n:110
Char: 62:e:101
Char: 63:.:46
Char: 64:
:13
Char: 65: :0
Char: 66:A:65
Char: 67:d:100
Char: 68:d:100
Char: 69:i:105
Char: 70:n:110
Char: 71:g:103
Char: 72: :32
Char: 73:a:97
Char: 74: :32
Char: 75:s:115
Char: 76:e:101
Char: 77:c:99
Char: 78:o:111
Char: 79:n:110
Char: 80:d:100
Char: 81: :32
Char: 82:l:108
Char: 83:i:105
Char: 84:n:110
Char: 85:e:101
Char: 86:
:13
Char: 87: :0

```

khoih-prog, 2020-08-06:
Oh, sorry, the same char(0), not char(10) or "\n" as you expect.

But I think that is correct because the end of the string is char(0).

"a" is stored as { 'a', char(0) }

We have to process the data when we read back from Flash if we know how it's stored.

AppsByDavideV, 2020-08-06:
Hi, I know about strings terminator, but it is not the case here. Sometimes
it writes correctly 10 instead of 0 and returns 10 reading it back. Surely
something inside the library at low level data processing...

inviato da XIAOMI

closed 2020-11-20 by khoih-prog