khoih-prog/WiFiNINA_Generic #6: Pinout for STM32F1 "blue pill"

Selective031, 2020-09-23:
Hi,

I have an issue with communicating with my "Adafruit atwinc1500 breakout board".
Its connected to a Blue Pill (STM32F103C8T6), using Arduino IDE 1.8.3 and Arduino Core 1.9.0
Pins:
MISO,MOSI,SCK (PA7,PA6,PA5)
PA4 - CS
PA3 - IRQ
PA2 - Reset
But I cannot communicate with the board.
I´m a little puzzled with the pinout found in the "Wifinina_pinout_generic.h" file.

Any help is very appreciated!

khoih-prog, 2020-09-23:
@tcpipchip

Can you offer some help here as I don't have the STM32F103C8T6.

tcpipchip, 2020-09-23:
hi @Selective031 

Here was my connection using my MCU with W102 (WIFININA)

See in right size the PINOUT of WIFININA

[image]

Selective031, 2020-09-23:
Hi @tcpipchip 

I´m using this schematic,
[image]

How would I translate this in wifinina_pinout_generic.h file ?

I tried to do it like this:
[image]

But I get no response from the board.

tcpipchip, 2020-09-23:
looks right!
Its the WIFININA ALIVE ?
Has the firmware inside ?
Can you measure on Osciloscope if there is answer ?

Selective031, 2020-09-23:
If I load the official WIFI101 (https://www.arduino.cc/en/Reference/WiFi101)
and use the "CheckWiFi101FirmwareVersion"
and add "WiFi.setPins(PA4,PA3,PA2)"
it will print out the firmware version, so it is working. However that library is not made for STM32 and will not work.

So my hope is WiFiNINA library ;)

khoih-prog, 2020-09-23:
@tcpipchip and @Selective031 

Just some info FYI. To use ATWINC1500 is a little bit tricky, and you have to experiment and explore yourself. It's difficult to help without the related hardware.

[ATWINC1500 Wiring-and-Test](https://learn.adafruit.com/adafruit-atwinc1500-wifi-module-breakout/wiring-and-test)

[image]

[image]

tcpipchip, 2020-09-23:
ATWINC1500, i had seen that you was using STM32

Selective031, 2020-09-23:
@khoih-prog Exactly, so when running this:

[image]

it will give me a reply in serial monitor:

[image]

khoih-prog, 2020-09-23:
@Selective031 @tcpipchip 

I currently **don't think ATWINC1500 is supported by WiFiNINA/WiFiNINA_Generic library** because there is no code to drive the IRQ pin which is actively used in ATWINC1500/WiFi101 library. Certainly just a good guess as I have no hardware to check.

### In WiFi101 lib:

#### 1. In WiFi.cpp

```cpp
void WiFiClass::setPins(int8_t cs, int8_t irq, int8_t rst, int8_t en)
{
  gi8Winc1501CsPin = cs;
  gi8Winc1501IntnPin = irq;
  gi8Winc1501ResetPin = rst;
  gi8Winc1501ChipEnPin = en;
}
```

#### 2. In bsp/source/nm_bsp_arduino.c

```cpp
int8_t gi8Winc1501CsPin = WINC1501_SPI_CS_PIN;
int8_t gi8Winc1501ResetPin = WINC1501_RESET_PIN;
int8_t gi8Winc1501IntnPin = WINC1501_INTN_PIN;
int8_t gi8Winc1501ChipEnPin = WINC1501_CHIP_EN_PIN;

...

void nm_bsp_register_isr(tpfNmBspIsr pfIsr)
{
	gpfIsr = pfIsr;
	attachInterruptMultiArch(gi8Winc1501IntnPin, chip_isr, FALLING);
}

...

void nm_bsp_interrupt_ctrl(uint8 u8Enable)
{
	if (u8Enable) {
		attachInterruptMultiArch(gi8Winc1501IntnPin, chip_isr, FALLING);
	} else {
		detachInterruptMultiArch(gi8Winc1501IntnPin);
	}
}

```

So it's better to use another shield for WiFiNINA library, such as Ublox W102, etc

tcpipchip, 2020-09-23:
W102 i know that works very fine on STM32, because i already tested!

Selective031, 2020-09-23:
Ok guys, thanks for trying!
You can close this.

closed 2020-09-23 by khoih-prog