khoih-prog/WiFiNINA_Generic #8: Invalid license?

bmcdonnell, 2021-05-21:
Thanks for your work on this library (and others!).

# Background

Per this library's README:

> This library is based on and modified from [Arduino WiFiNINA library](https://www.arduino.cc/en/Reference/WiFiNINA)

# Problem

The [WiFiNINA library for Arduino](https://github.com/arduino-libraries/WiFiNINA) is released under the GNU LGPL v2.1 or later, but this WiFiNINA_Generic library says it is released under the more permissive MIT license. As I understand it, this is not allowed for a derivative work, unless you've gotten permission from the copyright holder(s) of the original work.

# Solutions

I believe the easiest compliant solution would be to release WiFiNINA_Generic under LGPL v2.1 (and/or a later version). Alternatively, I think you'd need to get permission from the original work's copyright holder(s) to release it under a more permissive license (MIT, or whatever else you all agree to).

khoih-prog, 2021-05-21:
Hi @bmcdonnell 

I got similar issue with other library, as I really didn't spend precious time to look at the Licence, as in previous issue with another  library : [**Clarification on the license?** #8](https://github.com/khoih-prog/AsyncHTTPRequest_Generic/issues/8)

> I'm sorry I didn't spent time to look at the licenses and see the differences. It doesn't matter to me if it's GPLv3 or MIT, as long as users can use the library easily.
> 
> If there is issue, I'll convert back to GPLv3 without any problem at all.
> 
> Thanks for pointing out.

I certainly change the Licence shortly. I'd appreciate it if you can spot anywhere else.

Regards,

khoih-prog, 2021-05-21:
Done

closed 2021-05-21 by khoih-prog