khoih-prog/WiFiNINA_Generic #11: Wio Terminal can't communicate with module error

bfrenchrit, 2021-09-20:
Hi, I'm getting the following error when trying to run any of the examples:

Start WiFiNINA CheckFirmwareVersion on SAMD SEEED_WIO_TERMINAL
WiFiNINA_Generic v1.8.13
Communication with WiFi module failed!

I have flashed the latest firmware for the RTL8720 chip (2.1.3) could that be the problem?

khoih-prog, 2021-09-20:
**WIO Terminal** can't use WiFiNINA or [**WiFiNINA_Generic**](https://github.com/khoih-prog/WiFiNINA_Generic). 

You have to use [**Seeed_Arduino_rpcWiFi** Library](https://github.com/Seeed-Studio/Seeed_Arduino_rpcWiFi). 

These libraries of mine are supporting WIO_Terminal

1. [**WIOTerminal_WiFiManager**](https://github.com/khoih-prog/WIOTerminal_WiFiManager)
2. [**WebSockets_Generic**](https://github.com/khoih-prog/WebSockets_Generic)
3. [**SinricPro_Generic**](https://github.com/khoih-prog/SinricPro_Generic)

Go to [Seeedstudio Forum](https://forum.seeedstudio.com/) for more support.

Good Luck,

closed 2021-09-20 by khoih-prog

bfrenchrit, 2021-09-20:
Thanks!

Do those libraries support WPA2 Enterprise, I couldn’t find where they do.

Bryan French

Senior Lecturer and WMC Program Coordinator
School of Information
Golisano College of Computing and Information Sciences
Rochester Institute of Technology

[signature_897485934]Advisor RIT Players

Bldg 70-2619
152 Lomb Memorial Drive
Rochester, New York 14623

585-475-6511
***@***.***

khoih-prog, 2021-09-20:
All depends on the basic  [**Seeed_Arduino_rpcWiFi** Library](https://github.com/Seeed-Studio/Seeed_Arduino_rpcWiFi), on which those libraries rely on.

I haven't tried `WPA2 Enterprise` yet, but many other WiFi libraries, even ESP32, ESP8266, WiFiNINA, etc., can connect but also have certain issues with `WPA2 Enterprise`. You have to do more research on it.

It's better to use ESP32/ESP8266 if you have to use `WPA2 Enterprise` in University.