khoih-prog/WiFiNINA_Generic #15: Debug text comes out of the serial port when listening for UDP connections

NameOfTheDragon, 2021-11-16:
https://github.com/khoih-prog/WiFiNINA_Generic/blob/2bb99034a5238169f7d292a7c3f407733f4e7fff/src/utility/server_drv.cpp#L56

There is what looks like a rogue `#define` in the source that enables serial debugging. I just added a call to `UDP.parsePacket()` and it spews debugging diagnostics constantly.

Would you like me to submit a PR? (assuming I can figure out how to build the code).

khoih-prog, 2021-11-16:
Thanks. I'll remove it in master, which is a leftover from my previous debug. Sorry.

To remove and avoid unnecessary debugging messages, just change to

```
#define KH_WIFININA_SERVER_DRV_DEBUG   0
```

khoih-prog, 2021-11-16:
Actually, there are more leftovers you have to fix

1. [spi_drv.cpp#L66](https://github.com/khoih-prog/WiFiNINA_Generic/blob/master/src/utility/spi_drv.cpp#L66)

to be changed to 

```
#define KH_WIFININA_SPI_DEBUG   0
```
2. [WiFiUdp_Generic.cpp#L59](https://github.com/khoih-prog/WiFiNINA_Generic/blob/master/src/WiFiUdp_Generic.cpp#L59)

to be changed to 

```
#define KH_WIFININA_UDP_DEBUG   0
```

khoih-prog, 2021-11-16:
The master has been updated with those leftovers removed.

Thanks for your contribution.

closed 2021-11-16 by khoih-prog