khoih-prog/WiFiNINA_Generic #21: Issues with platformIO project getting error Communication with WiFi module failed!

gbernal, 2022-06-22:
### Describe the bug

A clear and concise description of what the bug is.

### Steps to Reproduce

Steps to reproduce the behavior. Including the [MRE](https://stackoverflow.com/help/minimal-reproducible-example) sketches

### Expected behavior

A clear and concise description of what you expected to happen.

### Actual behavior

A clear and concise description of what you expected to happen.

### Debug and AT-command log (if applicable)

A clear and concise description of what you expected to happen.

### Screenshots

If applicable, add screenshots to help explain your problem.

### Information

Please ensure to specify the following:

* Arduino IDE version (e.g. 1.8.13) or Platform.io version
* `ESP8266`,`ESP32` or `STM32` Core Version (e.g. ESP8266 core v2.7.4, ESP32 v1.0.5 or STM32 v1.9.0)
* Contextual information (e.g. what you were trying to achieve)
* Simplest possible steps to reproduce
* Anything that might be relevant in your opinion, such as:
  * Operating system (Windows, Ubuntu, etc.) and the output of `uname -a`
  * Network configuration
  
### Example

```
Arduino IDE version: 1.8.13
ESP32 Core Version 1.0.5
OS: Ubuntu 20.04 LTS
Linux xy-Inspiron-3593 5.4.0-66-generic #74-Ubuntu SMP Wed Jan 27 22:54:38 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

Context:
I encountered an endless loop while trying to connect to Local WiFi.

Steps to reproduce:
1. ...
2. ...
3. ...
4. ...
```

### Additional context

Add any other context about the problem here.

khoih-prog, 2022-06-22:
Hi @gbernal 

Hi @gbernal 

It's good that you went back and started over with the supported MKR1010. Be sure you have the updated [WiFi101](https://github.com/arduino-libraries/WiFi101) library and use Arduino WiFiNINA to test first to be sure the hardware is OK

Then use the latest **Arduino IDE v1.8.19**, and be sure everything is working OK with [**WiFiNINA_Generic**](https://github.com/khoih-prog/WiFiNINA_Generic), then with [**WiFiManager_NINA_Lite**](https://github.com/khoih-prog/WiFiManager_NINA_Lite).

**Moving to PIO only all is OK with Arduino IDE.**

---

> I have a custom board based on the sparkfun_samd51_micromod that has a Ublox W102 attached to it. I'm able to use the WiFiNINA from Arduino and Adafruit WiFiNINA Library by only changing the corresponding pins.

For currently unsupported `sparkfun_samd51_micromod`, this is so good and you can use this info to modify the [**WiFiNINA_Pinout_Generic.h**](https://github.com/khoih-prog/WiFiNINA_Generic/blob/master/src/WiFiNINA_Pinout_Generic.h) if necessary, then repeat the previous steps over again.

Sorry I can't help you any further as the  `sparkfun_samd51_micromod` and your hardware is so special and I don't have that kind of hardware combination.

---

As this is not an issue or bug of the library, I'm closing it now. Post for help on [**Discussions**](https://github.com/khoih-prog/WiFiNINA_Generic/discussions), Arduino Forum or  PIO Forum if PIO related issue if not a bug of this library.

Good Luck,

closed 2022-06-22 by khoih-prog

gbernal, 2022-06-22:
Dear @khoih-prog 
Thank you for responding so quickly. I have followed your suggestions and ran your library using the Arduino IDE as you can see in the screenshot attached. The hardware is working fine and runs as it should. 
[image]

I went back to PIO and created a fresh project and added the latest version of WifiNINA_Generic and W101 but I get the same issue as before. 

[image]

Thank you for your help in advance!

khoih-prog, 2022-06-22:
So please post your issue on PIO Forum for help as this is PIO bug / issue.